ros2/rviz #1227: Rviz2 crashes in ROS2 jazzy

rayee-github, 2024-07-09:
I installed ROS2 Jazzy on Ubuntu 24.04, but running rviz2 crashes immediately. The error message is shown in the attached image. This is an old issue. In the Foxy version, I encountered this problem when users switched from Xorg to Wayland. It could be simply resolved by using "QT_QPA_PLATFORM=xcb".

In the Humble version of rviz2, it seemed like the Wayland issue was resolved. Why has it reappeared in Jazzy, and why do users still have to manually input the command?

[image]

wushenhaoyu, 2024-07-12:
I have the same problem, I installed ROS rolling on Ubuntu 24.04. Frist time I came across the error, I thought it is the Raspberrt pi 's fault , I transform it to my vm ware. However, i came across the same error .

wushenhaoyu, 2024-07-12:
I am a rookie,Thank you for your description.

closed 2024-07-29 by rayee-github