ros2/rviz #1255: RViz2 Freezing with Autoware

dtonda8, 2024-08-05:
### Your environment
* OS Version: Ubuntu 22.04 (on UTM Apple Silicon)
* ROS Distro: Humble
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    (see launch.log file at the end)
    ```
* If source build, which git commit? (?) Followed the Default ROS2 installation (I think this is where I installed RViz), als using the [latest Autoware Source Installation](https://www.google.com/search?client=safari&rls=en&q=source+installation+autoware&ie=UTF-8&oe=UTF-8)
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: en_US.UTF-8
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`!

Rviz2 works find with an empty configuration. However, when following the [Autoware Planning Simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/#1-launch-autoware) instructions, The centre display freezes the second time I try to interact with the 3D space:

https://github.com/user-attachments/assets/3ae6de2e-c1e9-4745-bc8b-fe9da17ee063

`$HOME/.ros/log` output:  [launch.log](https://github.com/user-attachments/files/16491604/launch.log)

Yadunund, 2024-08-15:
Hello @dtonda8,

In this particular scenario, it would be great to isolate the cause of the issue since at the moment there are many confounding variables including an older version of ROS 2, 3rd Party extensions to ROS 2 (Autoware), the fact that this is all running within a VM, and lastly given that the VM itself is running on a tier-3 support OS for ROS 2.

It would greatly help if you could ascertain the following:
1. Whether the version of RViz using within Autoware is exactly the same as what's in this repo (humble branch) without any patches applied. If there are any modifications, please share the diff here.
2. If the issue can be reproduced with ROS 2 Rolling version of RViz on a Tier-1 OS outside of any VM, ie, this repo built from source off the latest `rolling` branch on Ubuntu 24.04. And specifically outside of Autoware. A simple self contained reproducible scenario would be excellent. 
3. Does running the above Autoware demo with ROS 2 Rolling make the problem go away?

Answering these would help us understand whether this is really a problem in RViz, Autoware, VM environment etc.

Personally I use RViz (ROS 2 Iron built from source) on M1 Mac running MacOS Sonoma 14.5 without any issues (No VM and not tested with Autoware).

dtonda8, 2025-05-09:
Hi @Yadunund, apologies for the egregiously late reply. It seems like I have buried this in the back of my backlog and switched to using a totally different computer.

I finally came back to this issue. The issue lies with the display you set on UTM. My issue above occurs whenever I am using a GPU display in UTM. Using a non-GPU supported display resolved my issue:

[image]

GPU supported displays work fine with rviz but doesn't work when using rviz + autoware/

closed 2025-05-09 by dtonda8