ros2/rviz #1256: `Tool::processKeyEvent` return value seems ignored

roncapat, 2024-08-07:
`int SelectionTool::processKeyEvent` return value is discarded in every call is made to it.

Qucik search: https://github.com/search?q=repo%3Aros2%2Frviz%20processKeyEvent&type=code

See, in some tools `Render` is returned but ignored. It doesn't seem to be an issue, however if we try to return something else (for example, `Finished`) the transition won't happen / the tool will not be disabled.

Example of `Render` being returned in current codebase:
https://github.com/ros2/rviz/blob/a72ef90031c25fceb0b9e58ce3fef58fc246584f/rviz_default_plugins/src/rviz_default_plugins/tools/select/selection_tool.cpp#L180-L190

I propose to handle the return value as it is done here for mouse event handler:
https://github.com/ros2/rviz/blob/a72ef90031c25fceb0b9e58ce3fef58fc246584f/rviz_common/src/rviz_common/visualization_manager.cpp#L623-L637

MichaelOrlov, 2024-08-15:
@ahcorde Could you please help to triage this issue?

roncapat, 2024-08-15:
Just for additional context, see #1257. It implements support for "Finished" flag in a back-portable way, while support for "Render" flag may be more complex to be introduced without breaking API.

roncapat, 2024-08-21:
I would keep this one open until the second PR to handle "Render" flag gets merged (I'll open it in a few days, sorry).

ahcorde, 2024-08-29:
Thank you for the patch @roncapat. Closing this issue

closed 2024-08-29 by ahcorde

roncapat, 2024-08-29:
You're welcome :)

PR #1257 (2024-08-21): Handle `Tool::Finished` returned by `processKeyEvent`
Partially fixes #1256.

This is a (probably backportable) patch that allows a Tool to declare its own use "Finished" and make Rviz re-select the default tool.

Please notice that some Tools defined in this codebase return another flag (`Render`, see attached issue) but this is not being handled too. 

### Note:

To process `Render` flag, an API breaking change shall be introduced (maybe in another PR) so that:
```
void ToolManager::handleChar(QKeyEvent * event, RenderPanel * panel)
```
becomes:
```
int ToolManager::handleChar(QKeyEvent * ev [... cut]
PR #1263 (2024-08-21): Handle `Tool::Finished` returned by `processKeyEvent` (backport #1257)
Partially fixes #1256.

This is a (probably backportable) patch that allows a Tool to declare its own use "Finished" and make Rviz re-select the default tool.

Please notice that some Tools defined in this codebase return another flag (`Render`, see attached issue) but this is not being handled too. 

### Note:

To process `Render` flag, an API breaking change shall be introduced (maybe in another PR) so that:
```
void ToolManager::handleChar(QKeyEvent * event, RenderPanel * panel)
```
becomes:
```
int ToolManager::handleCh [... cut]
PR #1264 (2024-08-21): Handle `Tool::Finished` returned by `processKeyEvent` (backport #1257)
Partially fixes #1256.

This is a (probably backportable) patch that allows a Tool to declare its own use "Finished" and make Rviz re-select the default tool.

Please notice that some Tools defined in this codebase return another flag (`Render`, see attached issue) but this is not being handled too. 

### Note:

To process `Render` flag, an API breaking change shall be introduced (maybe in another PR) so that:
```
void ToolManager::handleChar(QKeyEvent * event, RenderPanel * panel)
```
becomes:
```
int ToolManager::handleCh [... cut]
PR #1265 (2024-08-22): Handle `Tool::Finished` returned by `processKeyEvent` (backport #1257)
Partially fixes #1256.

This is a (probably backportable) patch that allows a Tool to declare its own use "Finished" and make Rviz re-select the default tool.

Please notice that some Tools defined in this codebase return another flag (`Render`, see attached issue) but this is not being handled too. 

### Note:

To process `Render` flag, an API breaking change shall be introduced (maybe in another PR) so that:
```
void ToolManager::handleChar(QKeyEvent * event, RenderPanel * panel)
```
becomes:
```
int ToolManager::handleCh [... cut]