ros2/rviz #1273: Consider the use of Backward-cpp or similar for getting stack traces during Segfaults

arjo129, 2024-09-02:
[`Backward.cpp`](https://github.com/bombela/backward-cpp) is already used fairly effectively by the Gazebo team.  It provides end users with a stack trace in the event that there is a segfault in their code by installing a signal handler. This can help us better understand issues with our own code and aid plugin developers with debugging as well.

Options to consider:
* Vendor it ourselves here.
* Depend on existing ros2 package: https://github.com/pal-robotics/backward_ros/tree/foxy-devel

clalancette, 2024-09-03:
So while I think this would be helpful, I think there are a few things we need to consider:

1.  Does Backward.cpp work across both Linux and Windows?
2.  If the answer to the previous question is "yes", can we get it from the platform (i.e. Ubuntu)?
3.  The signal handling story in ROS 2 is complicated.  If we want to implement this across the board, we'd probably have to at least implement it in rclcpp, rclpy, rosbag2, and possibly ros2cli and launch (the latter two are Python, but maybe we want to install it anyway).  We'd also have to integrate with the existing signal handling, which is quite complicated (especially on Windows).

arjo129, 2024-09-04:
> Does Backward.cpp work across both Linux and Windows?

Yes.

> If the answer to the previous question is "yes", can we get it from the platform (i.e. Ubuntu)?

Ubuntu seems to have a package but the package does not provide cmake targets. In the OSRF codebase it seems to be vendored in `osrf_testing_tools_cpp` and int 2 different places in gazebo.

> The signal handling story in ROS 2 is complicated. If we want to implement this across the board, we'd probably have to at least implement it in rclcpp, rclpy, rosbag2, and possibly ros2cli and [...]

I've never had luck with using backward-cpp with python. That being said, I think perusing the `rclcpp` repo there is an open issue https://github.com/ros2/rclcpp/issues/2396 which is related.