ros2/rviz #1277: rviz_rendering build fails on MacOS jazzy (Qt issue)

scott-robotics, 2024-09-09:
## Bug report

**Required Info:**

- Operating System:
  - MacOS 14.5 (M3)
- Installation type:
  - Source (jazzy)

#### Steps to reproduce issue

Follow the instructions for [iron](https://docs.ros.org/en/iron/Installation/Alternatives/macOS-Development-Setup.html) or [jazzy](https://docs.ros.org/en/jazzy/Installation/Alternatives/macOS-Development-Setup.html)

```
colcon build --merge-install --symlink-install --packages-up-to rviz_rendering
```

#### Expected behavior

Build succeeded

#### Actual behavior

Build failed

```
In file included from /Users/scottrobotics/Code/workspaces/ros2_jazzy/build/rviz_rendering/include/rviz_rendering/moc_render_window.cpp:10:
In file included from /Users/scottrobotics/Code/workspaces/ros2_jazzy/build/rviz_rendering/include/rviz_rendering/../../../../src/ros2/rviz/rviz_rendering/include/rviz_rendering/render_window.hpp:36:
In file included from /opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers/QObject:1:
/opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers/qobject.h:266:43: error: no template named 'QSlotObject' in namespace 'QtPrivate'
                           new QtPrivate::QSlotObject<Func2, typename QtPrivate::List_Left<typename SignalType::Arguments, SlotType::ArgumentCount>::Value,
                               ~~~~~~~~~~~^
```

#### Additional information

I can paste the full error log if useful. I'm also willing to help resolve this, just need to know where to poke

shenlebantongying, 2024-10-21:
I encountered the same error, and the reason appears to be some Qt headers from Qt installed from homebrew and Qt installed from the installer are conflicting.

I simply force uninstall the homebrew version and the problem is gone.

PS: I am an insanely random person from the internet. No idea what's this project, but Google bring me here.

senthilmanick, 2025-04-29:
Hit the same issue. Had qt 6.9 and qt 5.15. Both through Homebrew. 
brew unlink qt fixed it