ros2/rviz #1287: Rviz point cloud alpha flickering

dan11003, 2024-10-17:
Hi,
I recently migrated my code base to ros2 and is experience “flickering” for point clouds in Rviz2. It happens when using the alpha in PointCloud2 to improve visualization clarity. Occasionally, the effect of reduced alpha (increased transparency) vanishes, resulting in sudden very intense point clouds.
I.e. point cloud is still visualized but the alpha seems to flicker.
The decay time is set to 0.

Not sure if this relates to any other open Issue or not.

dan11003, 2024-10-17:
[image]
[image]

cottsay, 2024-11-01:
Interesting problem, thanks for filing a ticket.

It would be extremely helpful if we could reproduce this problem locally, and I think one of the easiest ways to do that would be if you could provide a bag file with some point cloud data that triggers the behavior.