ros2/rviz #1308: rviz2 running with low fps (5fps)

abdala-diasse, 2024-12-05:
I am currently running rviz2 with ros2 humble but my rviz is running very slowly compared to ros1 . With the same rviz config files rviz1 can run at 30fps but rviz2 can only run at 5pfs . 

Can anyone help me with this issue

Yadunund, 2024-12-09:
Please provide more information about your setup including
* OS version
* Whether you are using binaries or you've built ROS 2 Humble from source
* Link to a minimal repo with configs/instructions that allows us to reproduce the behavior

abdala-diasse, 2025-01-07:
I am running RViz ON Jetson  AGX 32 GB with JP 6.0
I have tried with both binaries and from source but  we still have the same issue

abdala-diasse, 2025-01-16:
any  feedback here

RENyunfan, 2025-01-18:
Hi @Yadunund @sloretz , I have met the similar problem and some tests are presented below

# rviz & rviz2 FPS Test

# 1. **Testing Environment**

- **Operating System**: Ubuntu 20.04
- **CPU**: 11th Gen Intel® Core™ i9-11900K @ 3.50GHz × 16
- **GPU**: NVIDIA GeForce RTX 3060/PCIe/SSE2 (NVIDIA Corporation)
- **Memory**: 62.6 GiB
- **OpenGL**: Version 3.6
- **ROS 1**: Noetic
- **Rviz 1**: Version 1.14.25
- **ROS 2**: Foxy
- **Rviz 2**: Latest version from [[this GitHub repository](https://github.com/ros2/rviz/tree/foxy)](https://github.com/ros2/rviz/tree/foxy)

# 2. **Problem Description**

When I attempt to publish a point cloud in both Rviz1 and Rviz2, I observe significant differences in their performance:

- **Rviz 1**: Very smooth with an FPS of up to 500 Hz (when adjusting the frame rate).
- **Rviz 2**: Becomes completely unresponsive when I try to change the view by dragging the mouse.

I have verified that both Rviz1 and Rviz2 are using the GPU, as shown in the following logs:

1. **Rviz 1 Output**:
    
    ```bash
    [ INFO] [1737213454.623461674]: rviz version 1.14.25
    [ INFO] [1737213454.623501868]: compiled against Qt version 5.12.8
    [ INFO] [1737213454.623509635]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1737213454.627971466]: Forcing OpenGl version 0.
    [ INFO] [1737213455.265363469]: Stereo is NOT SUPPORTED
    [ INFO] [1737213455.265403534]: OpenGL device: NVIDIA GeForce RTX 3060/PCIe/SSE2
    [ INFO] [1737213455.265414117]: OpenGl version: 4.6 (GLSL 4.6).
    ```
    
2. **Rviz 2 Output**: (The OpenGL Renderer device is obtained by slightly modify the source code)
    
    ```bash
    [INFO] [1737213896.198886037] [rviz2]: Stereo is NOT SUPPORTED
    [INFO] [1737213896.198984848] [rviz2]: OpenGL Vendor: NVIDIA Corporation
    OpenGL Renderer: NVIDIA GeForce RTX 3060/PCIe/SSE2
    OpenGL Version: 4.6.0 NVIDIA 470.256.02
    [INFO] [1737213896.198994700] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
    [INFO] [1737213896.275618870] [rviz2]: Stereo is NOT SUPPORTED
    
    ```
    
# 3. Reproduce the Issue
I have provided the necessary code and dataset to reproduce the issue on this [Notion page](https://renyunfan.notion.site/rviz-rviz2-FPS-Test-17f95176e07c80708869c2e07559767b?pvs=4). The resources include:

* Two Python scripts: one for ROS1 and one for ROS2
* A PCD file to serve as the dataset

I hope this helps in replicating the problem, and I look forward to your assistance.

Yadunund, 2025-01-23:
> I am running RViz2 on Jetson AGX 32 GB with JP 6.0
I have tried with both binaries and from source but we still have the same issue

@abdala-diasse Jetson AGX/JP6  is not a Tier-1 supported platform. [See](https://www.ros.org/reps/rep-2000.html#targeted-platforms). It is hard to isolate the cause of the problem (it could be a bug in RViZ, or display manager in JP, or..etc). It would be great if you could reproduce the issue on a Tier-1 platform (eg. Ubuntu) with the latest Rolling. Please share a minimal script to run that leads to the issue. 

@RENyunfan thanks for the additional details.

> ROS 2: Foxy

Foxy has long been end-of-life. Could you try your example with the latest Rolling or Jazzy?

> Rviz 2: Latest version from [[this GitHub repository](https://github.com/ros2/rviz/tree/foxy)](https://github.com/ros2/rviz/tree/foxy)

Please specify how you installed ROS 2. Whether from source or installed via binaries.