ros2/rviz #1314: Failed to create an OpenGL context. GLXBadDrawable

devvaibhav455, 2024-12-19:
ROS2 humble on Nvidia Jetson Orin NX (arm64) with JP 6.0 / Ubuntu 22.04

I have a startup script that launches a ROS2 launch file. One of the executables is rviz with a configuration that I select. Sometimes, I don't see anything on the screen.

I launch rviz2 using below

```
rviz = ExecuteProcess(
        cmd=[
            'ros2', 'run', 'rviz2', 'rviz2',
            '-d', str(filepath_config_rviz),
            '--ros-args',
            '--remap', '/tf:=/my_namespace/tf',
            '--remap', '/tf_static:=/my_namespace/tf_static',
        ],
        output='screen'
    )
```

Below is a screenshot of the problem.

[image]
[image]

```
[ros2-1] [INFO] [1734618134.390282224] [rviz2]: Stereo is NOT SUPPORTED
[ros2-1] [INFO] [1734618134.390545568] [rviz2]: OpenGL version: 4.6 (GLSL 4.6)
[ros2-1] [INFO] [1734618134.445473184] [rviz2]: Stereo is NOT SUPPORTED
[ros2-1] [ERROR] [1734618135.603434496] [rviz2]: Failed to create an OpenGL context. GLXBadDrawable
[ros2-1] [INFO] [1734618135.605991584] [rviz2]: Stereo is NOT SUPPORTED
[ros2-1] [ERROR] [1734618135.731887616] [rviz2]: Failed to create an OpenGL context. GLXBadDrawable
[ros2-1] [INFO] [1734618135.732704128] [rviz2]: Stereo is NOT SUPPORTED
```

I think this is the same as #908 but in my case, I don't face it in every boot.

mosfet80, 2024-12-21:
..try this patch #1305

devvaibhav455, 2024-12-25:
@mosfet80 Thanks for the quick patch. Should I just upgrade to [https://packages.ubuntu.com/oracular/libogre-1.12-dev](url)?