ros2/rviz #1319: Can't move a robot in gz sim using teleop twist keyboard

koa-ellie, 2024-12-28:
I already installed gz ros2 control.
My diff drive controller and joint broadcaster are already active. I can't still control the robot in gazebo.
Ubuntu 24.04.1
Jazzy Jalisco

Thanks!

koa-ellie, 2024-12-28:
I already remap the twist keyboard
ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -r cmd_vel:=/diff_drive_controller/cmd_vel

Still nothing happens

clalancette, 2024-12-28:
Please open questions like this on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

closed 2024-12-28 by clalancette

PranitVaikuntam, 2025-02-01:
I had the same problem. I saw that teleop_twist_keyboard was publishing to cmd_vel but it wasn't doing anything in gazebo. Then, I rand the command

Lukz0206, 2025-05-24:
I don't know if you still have the problem but I think it might be because unstamped twist messages aren't supported anymore. You need to send stamped twists. 
Maybe you can try this command 
ros2 run teleop_twist_keyboard teleop_twist_keyboard \
  --ros-args \
  -p stamped:=true \
  -r /cmd_vel:=/diff_drive_controller/cmd_vel