ros2/rviz #1320: Request FrameAligned view, as in ROS1

jowaibel, 2025-01-01:
In ROS1's rviz, there used to be a FrameAligned view that allowed to align the view exactly with a target frame. This is crucial for, e.g., a flight simulator where the user wants to see the view of the aircraft moving in 6 degrees of freedom. (In this case, the target frame would be the body frame.)

ThirdPersonFollower would work for a robot that moves on the horizontal plane only but not for an aircraft moving and rotating in 3 dimensions.

I cannot understand why this feature has not been ported and nobody asked for it since rviz2 came out. It would highly appreciate if it could be added to rviz2. Thank you very much!

christophebedard, 2025-01-09:
It was simply never ported. PRs are welcome :smile:

jowaibel, 2025-01-22:
Hi @christophebedard, I have not worked with the rviz source code before. If you could sketch what needs to be done to do the work and point me to the right fields, I will see what I can do.

christophebedard, 2025-01-24:
I've only looked at the RViz source code once myself, but any time is a good time to learn!

The plugin is defined in an XML file:

- In ROS 1, the `rviz/FrameAligned` plugin is defined here: https://github.com/ros-visualization/rviz/blob/250c0c2875758953f98f3b1982d11b55f527b295/plugin_description.xml#L294-L298
- In ROS 2, it would be added somewhere here: https://github.com/ros2/rviz/blob/7b31f1c421709e0e3c9e71446a474d959d90c571/rviz_default_plugins/plugins_description.xml#L450-L497

You can see that each plugin definition in the XML file has a `type`, which is essentially a class name. Each plugin class is registered in its `.cpp` file:

- In ROS 1, the `rviz::FrameViewController` class is registered here (see the `FrameViewController` class above): https://github.com/ros-visualization/rviz/blob/250c0c2875758953f98f3b1982d11b55f527b295/src/rviz/default_plugin/view_controllers/frame_view_controller.cpp#L165
- In ROS 2, for example, this is how the `rviz_default_plugins::view_controllers::FPSViewController` class is registered: https://github.com/ros2/rviz/blob/7b31f1c421709e0e3c9e71446a474d959d90c571/rviz_default_plugins/src/rviz_default_plugins/view_controllers/fps/fps_view_controller.cpp#L265-L266

Now, you can see that the ROS 1 `FrameViewController` class extends the `FPSViewController` class: https://github.com/ros-visualization/rviz/blob/250c0c2875758953f98f3b1982d11b55f527b295/src/rviz/default_plugin/view_controllers/frame_view_controller.h#L42. This `FPSViewController` class is itself [a plugin](https://github.com/ros-visualization/rviz/blob/250c0c2875758953f98f3b1982d11b55f527b295/plugin_description.xml#L284-L288), and it also exists in ROS 2, which I linked to above.

Therefore, I believe you just need to create a `FrameViewController` class that extends the existing `FPSViewController` class and essentially copy the logic from the ROS 1 implementation of `FrameViewController`. Looking at the ROS 1 `FPSViewController` class and the ROS 2 `FPSViewController` class, there are some differences, so you'll most likely have to adapt some things when porting `FrameViewController` to ROS 2.

Let me know if something isn't clear or if you have any questions.

closed 2026-10-30 by ahcorde