ros2/rviz #1332: Show pixel co-ordinate with their pixel value in image

devvaibhav455, 2025-01-24:
This is regarding a new feature request in Rviz2. Is it possible to show the current mouse pointer's pixel location and value just like OpenCV has it in Rviz. I think it will be easier to debug images in rosbags if someone is working with object detection tracking.

OpenCV shows information like this at the bottom of the image after it ends.

[image]

mjcarroll, 2025-01-27:
This would be an excellent feature, do you have any interest in working on it?

devvaibhav455, 2025-01-28:
@mjcarroll  Rviz code base will be new to me. If you or somebody could provide some guidance, that would shorten the learning curve and I could then take my time to implement and test it out. Please let me know how you feel about it.