ros2/rviz #1336: Text Marker spacing is disproportionately large for small text

rr-tom-noble, 2025-02-11:
**Version:**
humble binary (via apt)

**Issue:**

See attached screenshots. The first shows a Text Marker of scale 1. The screenshot below shows the same marker with a space, which also seems to display reasonably:

[image]

[image]

The following images show the text with a scale of 0.01. The space character (which is still just a single space, no changes made to the text content) ends up comically large compared to the rest of the text:

[image]

[image]

**Desired Solution:**

Spaces in text should also scale down with the text size. My feeling is that it may be using fixed size spacing (e.g. size 1 spacing regardless of scale). This desired solution is also the default behaviour in ROS1, so there may have been a fix already that just needs porting to ROS2

rr-tom-noble, 2025-02-12:
Leaving this open as I'm unsure if there's a wiki package similar to https://wiki.ros.org/rviz/DisplayTypes/Marker (which is ROS1, but is the only result I get). Is there an equivalent, and does it need updating to reflect this field? Happy to raise a PR if so.

audrow, 2025-02-20:
If anyone can do this, please port into the ROS 2 Rviz docs.