ros2/rviz #1338: Rviz2 [error] Vertex Program :rviz/glsl120/indexed_8bit_image.vert

BkaNmtien, 2025-02-18:
I using ROS2 Humble with WSL on window. I running rviz2 -d src/my_bot/config/main.rviz, then running slam_toolbox to import a available map
ros2 launch slam_toolbox online_async_launch.py params_file:=./src/my_bot/config/mapper_params_online_async.yaml use_sim_time:=true
Then display a error. please help:
QStandardPaths: wrong permissions on runtime directory /run/user/1000/, 0755 instead of 0700
[INFO] [1739853979.875756336] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1739853979.875846857] [rviz2]: OpenGl version: 4.1 (GLSL 4.1)
[INFO] [1739853979.904782952] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1739853980.056907971] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1739853984.240293902] [rviz2]: Trying to create a map of size 145 x 151 using 1 swatches
[ERROR] [1739853984.266594317] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
active samplers with a different type refer to the same texture image unit

BkaNmtien, 2025-02-19:
Please help!

PANJIACHENG-98, 2025-02-20:
have the problem been solved？i met the same problem。

BkaNmtien, 2025-02-24:
> have the problem been solved？i met the same problem。