ros2/rviz #1355: mapGetString Function Fails to Correctly Handle Missing Keys

zhihaoshang, 2025-03-29:
### Operating System:

Linux shangzh-VMware-Virtual-Platform 6.11.0-19-generic #19~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 17 11:51:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rviz

### 'ros2 doctor --report' output

```Formatted

```

### Steps to reproduce issue

## Description:
In the Config::mapGetString function, when the specified key is not found, the output parameter value_out is not assigned a default value. This leads to the value_out remaining uninitialized when the function is called, potentially causing undefined behavior or subsequent errors.
## Steps to Reproduce:
1.Call the mapGetString function with a non-existent key.

2.The function will return false, indicating that the key was not found, but will not assign any value to value_out.

3.If the caller does not check whether value_out is valid, unexpected results may occur.

## Write the following test case:
```cpp
#include <gtest/gtest.h>
#include <rviz_common/config.hpp>
TEST(Config, mapGetValue_key_not_found_null_check) {
  rviz_common::Config c;
  QString s;
  EXPECT_FALSE(c.mapGetString("non_existent_key", &s)); 
  EXPECT_EQ(s, "default_value");
}

```

## Output
```markdown
Running main() from /home/shangzh/ros2_jazzy/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from Config
[ RUN      ] Config.mapGetValue_key_not_found_null_check
/home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_common/test/config_test.cpp:7: Failure
Expected equality of these values:
  s
    Which is: {}
  "default_value"
    Which is: 0x6504765f44a0

[  FAILED  ] Config.mapGetValue_key_not_found_null_check (0 ms)
[----------] 1 test from Config (0 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (0 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] Config.mapGetValue_key_not_found_null_check

 1 FAILED TEST

```
## Tested Function Code:
```cpp
bool Config::mapGetString(const QString & key, QString * value_out) const
{
  QVariant v;
  if (mapGetValue(key, &v) && v.type() == QVariant::String) {
    *value_out = v.toString();
    return true;
  }
  return false;
}

```

## Suggested Fix:
```cpp
bool Config::mapGetString(const QString & key, QString * value_out) const
{
  QVariant v;
  if (mapGetValue(key, &v) && v.type() == QVariant::String) {
    *value_out = v.toString();
    return true;
  } else {
    *value_out = "default_value";  // Provide default value
    return false;
  }
}

```

### Expected behavior

When the specified key is not found, the mapGetString function should provide a default value for value_out. For example, it could be set to "default_value", so the caller receives consistent behavior whether the key exists or not.

### Actual behavior

When the key is not found, value_out remains uninitialized, which may lead to undefined behavior or errors in subsequent code.

zhihaoshang, 2025-04-02:
## Environment
OS Version: Ubuntu 24.04
Compiler name and version number: Ubuntu clang version 18.1.3
Source or binary build?
source build
build options: --mixin asan-gcc

ahcorde, 2025-04-03:
Closed by https://github.com/ros2/rviz/pull/1361

closed 2025-04-03 by ahcorde