ros2/rviz #1357: UniformStringStream::parseFloat Fails to Handle Invalid Float Formats Correctly

zhihaoshang, 2025-04-02:
### Operating System:

Linux shangzh-VMware-Virtual-Platform 6.11.0-19-generic #19~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 17 11:51:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rviz

### 'ros2 doctor --report' output

```Formatted

```

### Steps to reproduce issue

## Description
The UniformStringStream::parseFloat method does not correctly handle certain invalid float formats, such as "1.2.3" and "4,5,6". When the input contains multiple decimal points or commas, the function fails to set the failbit state properly, causing the program to incorrectly accept partially invalid inputs instead of returning an error.

## Steps to Reproduce
1.Run the following test for parseFloat:
```cpp
#include <gtest/gtest.h>
#include <locale>
#include "rviz_common/uniform_string_stream.hpp"
using namespace rviz_common;  // NOLINT
TEST(UniformStringStream, parse_floats_invalid_float_format) {
   UniformStringStream uss("1.2.3 4,5");
   float a, b;
   uss.parseFloat(a);
   uss.parseFloat(b);
   EXPECT_FALSE(!!uss);
}
int main(int argc, char **argv) {
   testing::InitGoogleTest(&argc, argv);
   return RUN_ALL_TESTS();
}
```

2.The test produces the following failure:
```markdown
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from UniformStringStream
[ RUN      ] UniformStringStream.parse_floats_invalid_float_format
/home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_common/test/uniform_string_stream_test.cpp:10: Failure
Value of: !!uss
  Actual: true
Expected: false

[  FAILED  ] UniformStringStream.parse_floats_invalid_float_format (0 ms)
[----------] 1 test from UniformStringStream (0 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (0 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] UniformStringStream.parse_floats_invalid_float_format

 1 FAILED TEST

```

### Expected behavior

If the input format is invalid (e.g., "1.2.3" or "4,5,6"), parseFloat should correctly set the failbit, making the input stream invalid.

After parsing failure, calling !!uss should return false, indicating an error in the input stream.

### Actual behavior

parseFloat incorrectly parses part of the input (e.g., "1.2.3" might be parsed as 1.2) and does not trigger failbit, causing the program to incorrectly assume that the input is valid.

### Additional information

## Root Cause
1.parseFloat extracts a substring from the stream and creates a new UniformStringStream float_reader(float_string); but does not check whether the entire string has been successfully parsed.

2.float_reader >> f; only reads the first valid part of float_string but fails to detect extra characters, allowing "1.2.3" to be incorrectly interpreted as 1.2 without an error.

## Suggested Fix
Modify parseFloat to ensure that the entire string is parsed correctly, preventing the acceptance of invalid inputs:
```cpp
void UniformStringStream::parseFloat(float & f)
{
  std::string float_string;
  *this >> float_string;
  
  size_t comma_index = float_string.find(',');
  if (comma_index != std::string::npos) {
    float_string[comma_index] = '.';
  }

  UniformStringStream float_reader(float_string);
  float_reader >> f;
  
  // **New check**: Ensure the entire string is correctly parsed
  if (float_reader.fail() || !float_reader.eof()) {
    this->setstate(std::ios::failbit);
  }
}

```

zhihaoshang, 2025-04-02:
## Environment
OS Version: Ubuntu 24.04
Compiler name and version number: Ubuntu clang version 18.1.3
Source or binary build?
source build
build options: --mixin asan-gcc

ahcorde, 2025-04-02:
PR https://github.com/ros2/rviz/pull/1360

closed 2025-04-02 by christophebedard