ros2/rviz #1380: ogreQuaternionAngularDistance Does Not Handle Non-Normalized Quaternions Properly

zhihaoshang, 2025-04-05:
### Operating System:

Linux shangzh-VMware-Virtual-Platform 6.11.0-19-generic #19~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 17 11:51:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rviz

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,LOOPBACK,RUNNING>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<UP,BROADCAST,RUNNING,MULTICAST>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.5, local=5.1.5
camera_info_manager                       : latest=5.1.5, local=5.1.5
camera_calibration_parsers                : latest=5.1.5, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
[... 395 lines cut ...]

### Expected behavior

The function should return the correct angular distance even when the input quaternions are not normalized, or it should explicitly enforce that the inputs must be unit quaternions (either via normalization or input validation).

### Actual behavior

The function does not normalize or validate the input quaternions before performing the computation. As a result, when non-unit quaternions are used, the calculated angle is incorrect. In the test case above, the result is π instead of the expected π/2.

### Additional information

##  Suggested Fixes
1.Automatically normalize inputs (recommended):
Add normalization inside the function to ensure valid input:
```cpp
first.normalise();
second.normalise();

```
2.Add input validation:
If normalization is avoided for performance reasons, assert or throw an error when non-normalized inputs are detected:
```cpp
assert(fabs(first.normalise() - 1.0f) < 1e-3 && "Input quaternion must be normalized.");

```

mjcarroll, 2025-04-05:
Thanks for the report, this does look like a bug, would you care to open a PR?

zhihaoshang, 2025-04-15:
> 感谢您的报告，这看起来确实是一个错误，您愿意打开 PR 吗？

Thank you for your response!

I’m glad to hear that you acknowledge it as a bug. I would appreciate it if you could submit the PR for the fix, as I’m currently not familiar with the process of submitting a PR myself. Your expertise in this area would ensure that the fix is implemented correctly and efficiently.

Please let me know if you need any additional information from my side to move forward with the PR.

Thank you again for your help!

zhihaoshang, 2025-04-30:
> Thanks for the report, this does look like a bug, would you care to open a PR?

Thank you for your patience! Since I'm not familiar with submitting PRs yet. Hopefully, developers who are more familiar with the project can help submit a PR for fixes.

MichaelOrlov, 2025-05-01:
@kscottz Help from the community is wanted here.

chen-harrison, 2025-06-02:
I actually unintentionally addressed this issue while fixing a different bug with that function. The commit is here: https://github.com/ros2/rviz/commit/656dcdd49e44f64d1f8b6f09a6d6f6effbdb896

@mjcarroll @MichaelOrlov I think this issue can probably be closed, unless an additional test case with non-normalized quaternions is wanted?

closed 2025-06-03 by ahcorde