ros2/rviz #1381: Precision Problem in ogreQuaternionAngularDistance with Nearly Identical Quaternions

zhihaoshang, 2025-04-05:
### Operating System:

Linux shangzh-VMware-Virtual-Platform 6.11.0-19-generic #19~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 17 11:51:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### ROS version or commit hash:

ros2 jazzy

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<LOOPBACK,UP,RUNNING>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<BROADCAST,MULTICAST,UP,RUNNING>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.5, local=5.1.5
camera_info_manager                       : latest=5.1.5, local=5.1.5
camera_calibration_parsers                : latest=5.1.5, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
turtlesim                                 : latest=1.8.3, local=1.8.3
trajectory_msgs                           : latest=5.3.6, local=5.3.5
tracetools_test                           : latest=8.2.3, local=8.2.3
tracetools_launch                         : latest=8.2.3, local=8.2.3
[... 389 lines cut ...]
 1 FAILED TEST

```

### Expected behavior

When the input quaternions represent almost the same rotation (e.g., difference on the order of 1e-6), the function should return an angular distance close to 0.0f, within a defined tolerance (e.g., 1e-6).

### Actual behavior

The current implementation returns a value of 2e-6, which exceeds the allowed error tolerance.

### Additional information

## Suggested Fix
We recommend replacing the current implementation with a more numerically stable approach using the dot product, as shown below:
```cpp
float ogreQuaternionAngularDistance(Ogre::Quaternion q1, Ogre::Quaternion q2)
{
  float dot = q1.Dot(q2);
  dot = std::clamp(dot, -1.0f, 1.0f);  // Prevent invalid input due to floating-point errors
  return 2.0f * acosf(fabs(dot));      // Use fabs to handle equivalent ±q cases
}

```

zhihaoshang, 2025-04-15:
> 感谢您的报告，这看起来确实是一个错误，您愿意打开 PR 吗？

Thank you for your response!

I’m glad to hear that you acknowledge it as a bug. I would appreciate it if you could submit the PR for the fix, as I’m currently not familiar with the process of submitting a PR myself. Your expertise in this area would ensure that the fix is implemented correctly and efficiently.

Please let me know if you need any additional information from my side to move forward with the PR.

Thank you again for your help!

zhihaoshang, 2025-04-30:
> 感谢您的报告，这看起来确实是一个错误，您愿意打开 PR 吗？

Thank you for your patience! Since I'm not familiar with submitting PRs. Hopefully, developers who are more familiar with the project can help submit a PR for fixes.

MichaelOrlov, 2025-05-01:
@kscottz Help from the community is wanted here.

Tanishq30052002, 2025-05-19:
@MichaelOrlov i will check this out

chen-harrison, 2025-05-20:
@Tanishq30052002 do you think I could work on this? I saw you responded to a couple issues here, and I'm trying to get into the realm of contributing to ROS/open source software and saw that this was tagged with 'good first issue'.

PR #1473 (2025-06-02): Fix Issue with Quaternion Angular Distance
## Description

Fixes #1381

### Is this user-facing behavior change?

### Did you use Generative AI?

### Additional Information

- Add new implementation for calculating angular distance between
  quaternions suggested by issue author, but include additional step of
  normalizing quaternions prior to the dot product operation
- Add gmock test to verify that new implementation works even with small
  angular distances, e.g. 1.0e-6
- Decrease tolerance on expected values of angle distance calculations
  in similar tests, for parity
- Tested on jazzy ( [... cut]