ros2/rviz #1389: Lack of Validity Check for Invalid Parameters in EffortVisual::EffortVisual Constructor

zhihaoshang, 2025-04-10:
### Operating System:

Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rviz

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<LOOPBACK,RUNNING,UP>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<BROADCAST,RUNNING,UP,MULTICAST>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.6, local=5.1.5
camera_info_manager                       : latest=5.1.6, local=5.1.5
camera_calibration_parsers                : latest=5.1.6, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
[... 446 lines cut ...]

When the EffortVisual constructor receives invalid parameters (e.g., nullptr), it should catch the invalid input and throw an appropriate exception, preventing the program from continuing with erroneous values.

### Actual behavior

When the EffortVisual constructor receives nullptr as either the scene_manager or parent_node parameter, it does not throw an exception, which leads to undefined behavior in the program later on.

### Additional information

## Suggested Fix
Add validity checks for the scene_manager and parent_node parameters in the EffortVisual::EffortVisual constructor. If either of these parameters is nullptr, an exception (such as std::invalid_argument or std::runtime_error) should be thrown to handle the invalid input properly. Below is the suggested fix:
```cpp
EffortVisual::EffortVisual(
  Ogre::SceneManager* scene_manager, Ogre::SceneNode* parent_node, float width, float scale)
  : scene_manager_(scene_manager), parent_node_(parent_node), width_(width), scale_(scale)
{
  if (!scene_manager) {
    throw std::invalid_argument("Invalid scene manager (nullptr provided).");
  }
  if (!parent_node) {
    throw std::invalid_argument("Invalid parent node (nullptr provided).");
  }
}

```

zhihaoshang, 2025-04-10:
## Test Case
```cpp
#include <gmock/gmock.h>
#include <OgreRoot.h>
#include <OgreSceneNode.h>
#include
#include "../ogre_testing_environment.hpp"
#include "../scene_graph_introspection.hpp"
#include "rviz_rendering/objects/effort_visual.hpp"
using namespace ::testing; // NOLINT
MATCHER_P(ColorEq, expected, "") {
return Ogre::Math::Abs(expected.a - arg.a) < 0.0001f &&
Ogre::Math::Abs(expected.r - arg.r) < 0.0001f &&
Ogre::Math::Abs(expected.g - arg.g) < 0.0001f &&
Ogre::Math::Abs(expected.b - arg.b) < 0.0001f;
}
MATCHER_P(Vector3Eq, expected, "") {
return Ogre::Math::Abs(expected.x - arg.x) < 0.0001f &&
Ogre::Math::Abs(expected.y - arg.y) < 0.0001f &&
Ogre::Math::Abs(expected.z - arg.z) < 0.0001f;
}
MATCHER_P(QuaterionEq, expected, "") {
return Ogre::Math::Abs(expected.x - arg.x) < 0.0001f &&
Ogre::Math::Abs(expected.y - arg.y) < 0.0001f &&
Ogre::Math::Abs(expected.z - arg.z) < 0.0001f &&
Ogre::Math::Abs(expected.w - arg.w) < 0.0001f;
}
class EffortVisualTestFixture : public ::testing::Test
{
protected:
void SetUp()
{
testing_environment_ = std::make_shared<rviz_rendering::OgreTestingEnvironment>();
testing_environment_->setUpOgreTestEnvironment();
}
std::shared_ptr<rviz_rendering::OgreTestingEnvironment> testing_environment_;
};
static Ogre::SceneNode * findForceArrow(Ogre::SceneNode * scene_node)
{
std::vector<Ogre::SceneNode *> arrows = rviz_rendering::findAllArrows(scene_node);
Ogre::BillboardChain * billboard_line = rviz_rendering::findOneBillboardChain(scene_node);
for (Ogre::SceneNode * arrow : arrows) {
if (billboard_line->getParentSceneNode()->getParent() == arrow->getParent()) {
return arrow;
}
}
return nullptr;
}

TEST_F(EffortVisualTestFixture, verify_no_memory_leak_on_exception) {
auto scene_manager = Ogre::Root::getSingletonPtr()->createSceneManager();
auto root_node = scene_manager->getRootSceneNode();
try {
auto effort_visual = std::make_shared<rviz_rendering::EffortVisual>(
nullptr, root_node, 0.0f, 0.0f); // Intentionally passing nullptr
FAIL() << "Exception should have been thrown due to invalid scene manager.";
} catch (const std::exception& e) {
SUCCEED();
}
// No other memory cleanup is needed as we are using smart pointers
}
```

## Output
```markdown
Running main() from gmock_main.cc
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from EffortVisualTestFixture
[ RUN      ] EffortVisualTestFixture.verify_no_memory_leak_on_exception
[rviz_rendering:debug] Available Renderers(1): OpenGL Rendering Subsystem, at /home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_rendering/src/rviz_rendering/render_system.cpp:289
[rviz_rendering:info] Stereo is NOT SUPPORTED, at /home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_rendering/src/rviz_rendering/render_system.cpp:531
[rviz_rendering:info] OpenGl version: 4.3 (GLSL 4.3), at /home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_rendering/src/rviz_rendering/render_system.cpp:272
/home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_rendering/test/rviz_rendering/objects/effort_visual_test.cpp:53: Failure
Failed
Exception should have been thrown due to invalid scene manager.

[  FAILED  ] EffortVisualTestFixture.verify_no_memory_leak_on_exception (1926 ms)
[----------] 1 test from EffortVisualTestFixture (1926 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (1926 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] EffortVisualTestFixture.verify_no_memory_leak_on_exception

 1 FAILED TEST

```

ahcorde, 2025-04-10:
PR https://github.com/ros2/rviz/pull/1395

closed 2025-04-11 by ahcorde