ros2/rviz #1392: calculateTotalDimensionsForPositioning Leading to Incorrect Bounding Box Calculation

zhihaoshang, 2025-04-10:
### Operating System:

Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rviz

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<LOOPBACK,UP,RUNNING>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<BROADCAST,MULTICAST,UP,RUNNING>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.6, local=5.1.5
camera_info_manager                       : latest=5.1.6, local=5.1.5
camera_calibration_parsers                : latest=5.1.6, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
[... 415 lines cut ...]
/home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_rendering/test/rviz_rendering/objects/movable_text_test.cpp:43: Failure
Value of: movable_text->getBoundingBox()
Expected: (has minimum (expected: Vector3(-2.11111, -2, 0))) and (has maximum (expected: Vector3(2.11111, 0, 0)))
  Actual: AxisAlignedBox(min=Vector3(-2, -2, 0), max=Vector3(2, 1.17549e-38, 1.17549e-38)) (of type Ogre::AxisAlignedBox)

[  FAILED  ] MovableTextTestFixture.centerAlignmentWorksCorrectlyWithSpaces (1922 ms)
[----------] 1 test from MovableTextTestFixture (1922 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (1922 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] MovableTextTestFixture.centerAlignmentWorksCorrectlyWithSpaces

 1 FAILED TEST

```

### Expected behavior

The calculateTotalDimensionsForPositioning function should correctly calculate the total height and width of the text, with the space character properly accounting for the expected width. The bounding box of the text should accurately reflect the content, especially when using H_CENTER horizontal alignment, where the bounding box should be symmetrically calculated and match the expected values.

### Actual behavior

The actual behavior shows a discrepancy in the calculation of space width, leading to incorrect total width and, consequently, an incorrect bounding box. The test assertions expect certain values for the bounding box, but the actual max value is unexpectedly small, particularly when dealing with space characters.

MichaelOrlov, 2025-05-01:
@kscottz Help from the community is wanted here.

Tanishq30052002, 2025-05-19:
@MichaelOrlov i will check this out

Tanishq30052002, 2025-05-20:
@zhihaoshang unable to replicate the error
Anything which you missed to share?

I am using the ubuntu 22.04 jazzy branch

zhihaoshang, 2025-05-20:
> 无法复制错误 Anything which you missed to share？
> 
> 我正在使用 ubuntu 22.04 jazzy 分支

Environment
OS Version: Ubuntu 24.04
rviz version: ros2 jazzy
Compiler name and version number: Ubuntu clang version 18.1.3
Source or binary build?
source build
build options: --mixin asan-gcc

Tanishq30052002, 2025-05-20:
> > 无法复制错误 Anything which you missed to share？
> > 我正在使用 ubuntu 22.04 jazzy 分支
> 
> Environment OS Version: Ubuntu 24.04 rviz version: ros2 jazzy Compiler name and version number: Ubuntu clang version 18.1.3 Source or binary build? source build build options: --mixin asan-gcc

I will try with 22.04 once