ros2/rviz #1405: WrenchVisual::setForceColor Function Fails to Validate Color Range

zhihaoshang, 2025-04-12:
### Operating System:

```Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

ros2 jazzy

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<LOOPBACK,UP,RUNNING>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<BROADCAST,UP,MULTICAST,RUNNING>
mtu          : 1500
broadcast    : 192.168.148.255

   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.11.0-21-generic-x86_64-with-glibc2.39
release          : 6.11.0-21-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0

```
</details>

### Steps to reproduce issue

## Environment
OS Version: Ubuntu 24.04
rviz version: ros2 jazzy
Compiler name and version number: Ubuntu clang version 18.1.3
Source or binary build?
source build
build options: --mixin asan-gcc

## Description
During unit testing, the WrenchVisual::setForceColor function fails to correctly validate the input color range. According to the function implementation, the values for r, g, b, and a should be restricted to the range [0.0, 1.0]. However, when the input values exceed this range, the function does not throw an exception, leading to the failure of the test case.

## Test Case
```cpp
#include <gmock/gmock.h>
#include <memory>
#include <string>
#include <vector>
#include <OgreBillboardChain.h>
#include <OgreManualObject.h>
#include <OgreRoot.h>
#include <OgreSceneNode.h>
#include "../ogre_testing_environment.hpp"
#include "../scene_graph_introspection.hpp"
#include "rviz_rendering/objects/wrench_visual.hpp"
#include "rviz_rendering/objects/billboard_line.hpp"
#include "rviz_rendering/objects/arrow.hpp"
using namespace ::testing;  // NOLINT
MATCHER_P(Vector3Eq, expected, "") {
  return Ogre::Math::Abs(expected.x - arg.x) < 0.0001f &&
         Ogre::Math::Abs(expected.y - arg.y) < 0.0001f &&
         Ogre::Math::Abs(expected.z - arg.z) < 0.0001f;
}
class WrenchVisualTestFixture : public ::testing::Test
{
protected:
  void SetUp()
  {
    testing_environment_ = std::make_shared<rviz_rendering::OgreTestingEnvironment>();
    testing_environment_->setUpOgreTestEnvironment();
  }
  std::shared_ptr<rviz_rendering::OgreTestingEnvironment> testing_environment_;
};

Ogre::SceneNode * findForceArrow(Ogre::SceneNode * scene_node)
{
  auto arrows = rviz_rendering::findAllArrows(scene_node);
  auto billboard_line = rviz_rendering::findOneBillboardChain(scene_node);
  for (const auto & arrow : arrows) {
    if (billboard_line->getParentSceneNode()->getParent() != arrow->getParent()) {
      return arrow;
    }
  }
  return nullptr;
}
TEST_F(WrenchVisualTestFixture, setForceColor_validates_color_range) {
  auto scene_manager = Ogre::Root::getSingletonPtr()->createSceneManager();
  auto root_node = scene_manager->getRootSceneNode();
  auto wrench_visual = std::make_shared<rviz_rendering::WrenchVisual>(scene_manager, root_node);

  // Valid range
  EXPECT_NO_THROW(wrench_visual->setForceColor(0.0f, 0.0f, 0.0f, 0.0f));
  EXPECT_NO_THROW(wrench_visual->setForceColor(1.0f, 1.0f, 1.0f, 1.0f));

  // Invalid range
  EXPECT_THROW(wrench_visual->setForceColor(-0.1f, 0.5f, 0.5f, 1.0f), std::invalid_argument);
  EXPECT_THROW(wrench_visual->setForceColor(0.5f, 1.1f, 0.5f, 1.0f), std::invalid_argument);
}
```

## Output
```markdown
Running main() from gmock_main.cc
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from WrenchVisualTestFixture
[ RUN      ] WrenchVisualTestFixture.setForceColor_validates_color_range
[rviz_rendering:debug] Available Renderers(1): OpenGL Rendering Subsystem, at /home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_rendering/src/rviz_rendering/render_system.cpp:289
[rviz_rendering:info] Stereo is NOT SUPPORTED, at /home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_rendering/src/rviz_rendering/render_system.cpp:531
[rviz_rendering:info] OpenGl version: 4.3 (GLSL 4.3), at /home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_rendering/src/rviz_rendering/render_system.cpp:272
/home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_rendering/test/rviz_rendering/objects/wrench_visual_test.cpp:53: Failure
Expected: wrench_visual->setForceColor(-0.1f, 0.5f, 0.5f, 1.0f) throws an exception of type std::invalid_argument.
  Actual: it throws nothing.

/home/shangzh/ros2_jazzy/src/ros2/rviz/rviz_rendering/test/rviz_rendering/objects/wrench_visual_test.cpp:54: Failure
Expected: wrench_visual->setForceColor(0.5f, 1.1f, 0.5f, 1.0f) throws an exception of type std::invalid_argument.
  Actual: it throws nothing.

[  FAILED  ] WrenchVisualTestFixture.setForceColor_validates_color_range (1879 ms)
[----------] 1 test from WrenchVisualTestFixture (1879 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (1879 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] WrenchVisualTestFixture.setForceColor_validates_color_range

 1 FAILED TEST
```

### Expected behavior

When the setForceColor function receives invalid color values (e.g., values for r, g, b, or a less than 0.0 or greater than 1.0), it should throw a std::invalid_argument exception.

Specifically:

When the color values are like -0.1 or 1.1, it should throw a std::invalid_argument exception.

Only when the color values are within the range [0.0, 1.0], should the function accept the color values.

### Actual behavior

The function does not throw any exception, even when the color values are out of range (e.g., -0.1 or 1.1).

This leads to the failure of the unit test setForceColor_validates_color_range, which does not meet the expected exception-throwing behavior.

### Additional information

## Suggested Fix
Add range validation: Include a check for the valid range of color values in the setForceColor function. If any color value is less than 0.0 or greater than 1.0, throw a std::invalid_argument exception.
```cpp
void WrenchVisual::setForceColor(float r, float g, float b, float a) {
    if (r < 0.0f || r > 1.0f || g < 0.0f || g > 1.0f || b < 0.0f || b > 1.0f || a < 0.0f || a > 1.0f) {
        throw std::invalid_argument("Color values must be in the range [0.0, 1.0]");
    }
    arrow_force_->setColor(r, g, b, a);
}
```

ahcorde, 2025-04-14:
Not really sure if we should throw an exception because a color is wrong, I will be happy just clamping the value between the right range.

zhihaoshang, 2025-04-15:
> 不确定我们是否应该因为颜色错误而引发异常，我很乐意将值限制在正确的范围内。

Thank you for your response!

I understand your point about clamping the values instead of throwing an exception. However, from a testing and debugging perspective, silently clamping out-of-range color values can make it harder to detect incorrect usage or logic errors in the calling code. If developers mistakenly pass invalid color values, it is often more helpful to fail fast with a clear exception, rather than allowing the program to continue with unintended visual outputs.

In our use case, strict input validation is important because it ensures that the API behaves predictably and that incorrect usage is caught early during development. This is especially critical in environments like robotics and visualization tools, where small errors in visual representation may mask deeper underlying issues.

Therefore, I would kindly suggest adding a check that throws a std::invalid_argument when the input values are outside the valid [0.0, 1.0] range. This would help ensure more robust and maintainable code. I would greatly appreciate it if you could submit a PR to address this issue. Please let me know if I can assist in any way with the process.

Thank you again for considering this!

ahcorde, 2025-04-25:
https://github.com/ros2/rviz/pull/1437

closed 2025-04-29 by ahcorde