ros2/rviz #1407: splitStringIntoTrimmedItems may crash when processing items consisting solely of whitespace

zhihaoshang, 2025-04-12:
### Operating System:

```Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rviz

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,RUNNING,LOOPBACK>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<BROADCAST,MULTICAST,UP,RUNNING>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.6, local=5.1.5
camera_info_manager                       : latest=5.1.6, local=5.1.5
camera_calibration_parsers                : latest=5.1.6, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
[... 431 lines cut ...]

For input like "a, \n, b", the function attempts to trim the whitespace-only item "\n". Since the item becomes empty after trimming, calling erase(whitespace_back.base(), item.end()) and erase(item.begin(), whitespace_front) may result in invalid iterator operations and crash the program at runtime.

The issue can be detected with AddressSanitizer, producing an error similar to:
```cpp
AddressSanitizer: heap-buffer-overflow on address ...
READ of size 1 at 0x...
```

### Additional information

## Suggested Fix
Before performing trimming operations with find_if_not, add a check to ensure the item is not entirely composed of whitespace:
```cpp
if (std::all_of(item.begin(), item.end(), [](char character) {
      return std::isspace<char>(character, std::locale(""));
    })) {
  item.clear();
} else {
  auto whitespace_front = std::find_if_not(...);
  auto whitespace_back = std::find_if_not(...);
  item.erase(whitespace_back.base(), item.end());
  item.erase(item.begin(), whitespace_front);
}
```

zhihaoshang, 2025-04-12:
## Description
When calling the splitStringIntoTrimmedItems function with a string that contains only delimiters or spaces, a memory access error occurs. The issue arises because the std::find_if_not function may return an invalid iterator when no non-whitespace characters are found, leading to an out-of-bounds access during the subsequent erase operation.

## Test Case
```cpp
#include <string>
#include <vector>
#include "gtest/gtest.h"
#include "gmock/gmock.h"
#include "../../src/rviz_rendering/string_helper.cpp"  // NOLINT (build/include)
using namespace ::testing;  // NOLINT

TEST(String_Helper__Test, HandlesNothingBesidesDelimiters) {
    std::vector<std::string> expected{};
    std::string test_string(",,\n,\n,,");
    std::vector<std::string> actual = rviz_rendering::string_helper::splitStringIntoTrimmedItems(
        test_string, ',');
    ASSERT_THAT(expected, Eq(actual));
}
```

## Output
```markdown
Running main() from gmock_main.cc
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from String_Helper__Test
[ RUN      ] String_Helper__Test.HandlesNothingBesidesDelimiters
=================================================================
==236015==ERROR: AddressSanitizer: negative-size-param: (size=-1)
    #0 0x74d9b78fafb6 in memmove ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors_memintrinsics.inc:98
    #1 0x74d9b536883f in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_erase(unsigned long, unsigned long) (/lib/x86_64-linux-gnu/libstdc++.so.6+0x16883f) (BuildId: ca77dae775ec87540acd7218fa990c40d1c94ab1)
    #2 0x74d9b53693ca in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::erase(__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, __gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >) (/lib/x86_64-linux-gnu/libstdc++.so.6+0x1693ca) (BuildId: ca77dae775ec87540acd7218fa990c40d1c94ab1)
    #3 0x5c83750eecac in rviz_rendering::string_helper::splitStringIntoTrimmedItems(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char) (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0x25cac) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #4 0x5c83750ef2de in String_Helper__Test_HandlesNothingBesidesDelimiters_Test::TestBody() (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0x262de) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #5 0x5c8375194ff1 in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0xcbff1) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #6 0x5c8375182ebb in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0xb9ebb) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #7 0x5c837512a303 in testing::Test::Run() (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0x61303) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #8 0x5c837512badf in testing::TestInfo::Run() (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0x62adf) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #9 0x5c837512ce38 in testing::TestSuite::Run() (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0x63e38) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #10 0x5c8375153b63 in testing::internal::UnitTestImpl::RunAllTests() (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0x8ab63) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #11 0x5c837519844c in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0xcf44c) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #12 0x5c8375186158 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0xbd158) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #13 0x5c837515016d in testing::UnitTest::Run() (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0x8716d) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #14 0x5c83750fa60e in RUN_ALL_TESTS() (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0x3160e) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #15 0x5c83750fa55a in main (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0x3155a) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)
    #16 0x74d9b4e2a1c9 in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #17 0x74d9b4e2a28a in __libc_start_main_impl ../csu/libc-start.c:360
    #18 0x5c83750ee314 in _start (/home/shangzh/ros2_jazzy/build/rviz_rendering/string_helper_test+0x25314) (BuildId: 588a977c4d4f370026afe2cbca5e7ef6a5e3c55c)

[... cut ...]
    return std::isspace<char>(character, std::locale(""));
  });
auto whitespace_back = std::find_if_not(
  item.rbegin(), item.rend(), [](char character) {
    return std::isspace<char>(character, std::locale(""));
  });

// Only perform erase operation if the string is not empty
if (whitespace_front != item.end() && whitespace_back != item.rend()) {
  item.erase(whitespace_back.base(), item.end());
  item.erase(item.begin(), whitespace_front);
}
if (!item.empty()) {
  filenames.push_back(item);
}
```

ahcorde, 2025-04-14:
https://github.com/ros2/rviz/pull/1412

closed 2025-04-14 by ahcorde