ros2/rviz #1409: splitStringIntoTrimmedItems unexpectedly drops empty tokens during string splitting, inconsistent with common split semantics

zhihaoshang, 2025-04-12:
### Operating System:

```Linux shangzh-VMware-Virtual-Platform 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

ros2 jazzy

### Client library (if applicable):

rviz

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<RUNNING,UP,LOOPBACK>
mtu          : 65536
inet         : 192.168.148.137
inet4        : ['192.168.148.137']
ether        : 00:0c:29:be:c8:19
inet6        : ['fe80::20c:29ff:febe:c819%ens33']
netmask      : 255.255.255.0
device       : ens33
flags        : 4163<MULTICAST,RUNNING,BROADCAST,UP>
mtu          : 1500
broadcast    : 192.168.148.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.7
rviz2                                     : latest=14.1.8, local=14.1.7
rviz_default_plugins                      : latest=14.1.8, local=14.1.7
rviz_common                               : latest=14.1.8, local=14.1.7
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.4
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
image_common                              : latest=5.1.6, local=5.1.5
camera_info_manager                       : latest=5.1.6, local=5.1.5
camera_calibration_parsers                : latest=5.1.6, local=5.1.5
yaml_cpp_vendor                           : latest=9.0.1, local=9.0.1
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.5
visualization_msgs                        : latest=5.3.6, local=5.3.5
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
[... 404 lines cut ...]
```
The expected result is a vector of 1,000,001 empty strings, consistent with the behavior of standard split functions.

### Actual behavior

The returned result is an empty vector:
```cpp
result.size() == 0
```
All split tokens, which are empty strings in this case, are filtered out without warning. This behavior is not documented and is likely to be surprising to users.

### Additional information

## Suggested Fix
We recommend one or more of the following improvements:

1.Adjust the behavior: Remove the filtering condition so that empty tokens are preserved.
```cpp
// Remove `if (!item.empty())` check
filenames.push_back(item);
```

2.Add a configurable parameter: Introduce a boolean parameter such as keep_empty = true to give users control over whether to retain empty tokens.

3.Improve documentation: If the current behavior is intentional, update the function documentation to clearly state that empty strings will be discarded.

InvincibleRMC, 2025-04-25:
@zhihaoshang it seems to me `string_helper::splitStringIntoTrimmedItems` seems to be functioning to design. Rather than modifying the function could you write your own `split` function to act more like Python's `split`?

kscottz, 2025-04-28:
@InvincibleRMC we believe the contributor in this case is using some sort of combination of AI and a fuzzer / static analyzer. In past issues they've said they can't really provide details. If you have given the issue a hard look and don't see any problem we can probably close the ticket, just LMK.

InvincibleRMC, 2025-04-28:
It seems to me that this isn't a bug and an intentional design decision instead.

kscottz, 2025-04-28:
@InvincibleRMC I'll take your word for it and close the issue.

closed 2025-04-28 by kscottz