ros2/rviz #1454: RVIZ2 executable aborts opening due to TF icon file name mismatch

pradyum, 2025-05-02:
### Operating System:

```Linux pc-QEMU-Virtual-Machine 6.11.0-25-generic #25~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 15 15:35:44 UTC 2 aarch64 aarch64 aarch64 GNU/Linux```

### ROS version or commit hash:

kilted

### RMW implementation (if applicable):

fastrtsp, cyclone, zenoh

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : UP,LOOPBACK,RUNNING
mtu          : 65536
inet         : 10.0.1.230
inet4        : ['10.0.1.230']
ether        : 2a:dd:b1:59:97:a6
inet6        : ['fdef:6def:9c40:0:feb1:d57e:de30:6f8f', 'fdef:6def:9c40:0:28dd:b1ff:fe59:97a6', 'fe80::28dd:b1ff:fe59:97a6%enp0s1']
netmask      : 255.255.255.0
device       : enp0s1
flags        : UP,BROADCAST,RUNNING,MULTICAST
mtu          : 1500
broadcast    : 10.0.1.255

   PACKAGE VERSIONS
rviz2                                     : latest=15.0.0, local=15.0.0
rqt_reconfigure                           : latest=1.7.0, local=1.7.0
launch_ros                                : latest=0.28.1, local=0.28.1
message_filters                           : latest=7.1.0, local=7.1.0
sdl2_vendor                               : latest=3.3.0, local=3.3.0
actionlib_msgs                            : latest=5.5.0, local=5.5.0
examples_rclpy_executors                  : latest=0.20.5, local=0.20.5
pcl_msgs                                  : latest=1.0.0, local=1.0.0
ament_cmake_copyright                     : latest=0.19.2, local=0.19.2
uncrustify_vendor                         : latest=3.1.0, local=3.1.0
examples_rclcpp_multithreaded_executor    : latest=0.20.5, local=0.20.5
ros_core                                  : latest=0.12.0, local=0.12.0
ament_lint_common                         : latest=0.19.2, local=0.19.2
unique_identifier_msgs                    : latest=2.7.0, local=2.7.0
tf2_tools                                 : latest=0.41.0, local=0.41.0
rosidl_generator_py                       : latest=0.24.1, local=0.24.1
examples_rclcpp_minimal_service           : latest=0.20.5, local=0.20.5
rqt_publisher                             : latest=1.9.0, local=1.9.0
ros2multicast                             : latest=0.38.0, local=0.38.0
rcl_yaml_param_parser                     : latest=10.1.0, local=10.1.0
launch                                    : latest=3.8.1, local=3.8.1
ament_cmake_lint_cmake                    : latest=0.19.2, local=0.19.2
qt_gui                                    : latest=2.9.1, local=2.9.1
rqt_bag_plugins                           : latest=2.0.2, local=2.0.2
depthimage_to_laserscan                   : latest=2.5.1, local=2.5.1
examples_rclcpp_minimal_action_server     : latest=0.20.5, local=0.20.5
ament_xmllint                             : latest=0.19.2, local=0.19.2
sros2                                     : latest=0.15.2, local=0.15.2
std_msgs                                  : latest=5.5.0, local=5.5.0
ros2service                               : latest=0.38.0, local=0.38.0
ros2doctor                                : latest=0.38.0, local=0.38.0
ament_cmake_export_link_flags             : latest=2.7.3, local=2.7.3
image_transport                           : latest=6.1.1, local=6.1.1
laser_geometry                            : latest=2.10.0, local=2.10.0
[... 285 lines cut ...]

or

```~$ rviz2```

### Expected behavior

**rviz2 window has to be opened**

### Actual behavior

aborts with error message:

```
terminate called after throwing an instance of 'resource_retriever::Exception'
  what():  Error retrieving file [/opt/ros/kilted/share/rviz_default_plugins/icons/classes/TF.svg]: Failed to open file
Aborted (core dumped)
```

### Additional information

The TF icon in the classes is of type PNG
> https://github.com/ros2/rviz/blob/cfa75dbec321d40e50f75d5c414b331537fe288e/rviz_default_plugins/icons/classes/TF.png

But, the executable is trying to open the TF icon of type SVG which makes the executable to abort.

pradyum, 2025-05-02:
This issue is observed in 

ros2/kilted_tutorial_party#483
ros2/kilted_tutorial_party#476
ros2/kilted_tutorial_party#470

christophebedard, 2025-05-03:
Should be fixed by #1455

pradyum, 2025-05-05:
@ahcorde Tested on Ubuntu arm64 - The Rviz is **working** fine. 

[image]