ros2/rviz #1468: Better handling of missing transport plugins

LuAPi, 2025-05-23:
I'm not sure whether this should be a bug report or a feature request.

On Ubuntu 24.04, jazzy rviz2 offers the option to subscribe to an image topic as a camera view using compressed transport. It still offers this option even if `ros-jazzy-compressed-image-transport` is not installed. Selecting this and trying to add the display results in rviz freezing and then closing from a sigabrt.

This could be seen as a bug since a user doesn't expect rviz to suddenly crash with no warning when adding a topic to be displayed, or to be offered to subscribe to a compressed topic if they lack the packages to do so.

It could also be viewed as not a bug because "print an error message and die" is a common type of error handling approach; and indeed when one does this there is a message left in the terminal to explain everything that one can find after the Ubuntu crash reporter has gone away:
```
terminate called after throwing an instance of 'image_transport::TransportLoadException'
  what():  Unable to load plugin for transport 'image_transport/compressed_sub', error string:
According to the loaded plugin descriptions the class image_transport/compressed_sub with base class type image_transport::SubscriberPlugin does not exist. Declared types are  image_transport/raw_sub
Aborted (core dumped)
```

I'd like to make some suggestions that might improve the user experience. The first would be to not display the option to use compressed forms of transport if the required plugins are missing. The second would be to popup an alert message box to report the issue to the user and to cancel adding the topic display rather than killing rviz.

ahcorde, 2025-06-19:
Do you mind to try this patch https://github.com/ros2/rviz/pull/1488

LuAPi, 2025-07-10:
Apologies for the delay.
It works. 👍